arduino/Arduino #11311: [Library Manager] Change of Library Name

mrfaptastic, 2021-02-28:
Library: https://github.com/mrfaptastic/Easy-IoT-Arduino-CC1101-LORA

From: "Easy ESP Arduino IoT with CC1101"
To: "Easy IoT with CC1101 (Low-Power Sub-1GHz LORA-like)"

per1234, 2021-02-28:
Hello again @mrfaptastic. There are restrictions on which characters are allowed in the library.properties `name` value. The characters `(` and `)` in your currently proposed name are prohibited. You can see the rules for the name here:
https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

After doing that, you'll need to make a new release or tag.

Once everything is in compliance with the library specification and [FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-change-my-librarys-name), please comment here to let us know we can proceed.

mrfaptastic, 2021-02-28:
Thanks again per1234. Please change name to: "Easy IoT with CC1101 - Sub-1GHz LORA-like"

I have updated the library files and created a new version in the respository.

cmaglie, 2021-03-03:
Done.

closed 2021-03-03 by cmaglie