arduino/Arduino #11312: Arduino 1.8.13 on Mac OS Big Sur : compile fails if space in path of .ino file location

xavbart, 2021-02-28:
As reported here:
https://forum.arduino.cc/index.php?topic=728254.0#main_content_section
and confirmed by my tests, 
if the path to the .ino file has one or many spaces in the name of the _directories_ enclosing it (no issue with spaces in the .ino filename), the compiler fails with error 64.
The error is actually quite transparent:
`bash: line 0: [: too many arguments
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file ... target_directory
exit status 64
/Applications/Arduino.app/Contents/Java/arduino-builder returned 64
Error compiling for board ESP32 Dev Module.`
I guess the argument giving the path to the file to compile should be properly quoted.

xavbart, 2021-02-28:
Arduino version 1.8.13 on Mac OS Big Sur Beta 11.3 Beta (20E5186d)

per1234, 2021-02-28:
Hi @xavbart. Thanks for your report. I see this is an error specific to the 3rd party esp32 boards platform:
https://github.com/espressif/arduino-esp32/issues/4579
Arduino does not maintain that software, so I'll go ahead and close this as off topic for this issue tracker. You may want to subscribe to https://github.com/espressif/arduino-esp32/issues/4579 to get updates on the progress of the fix for this issue.

I see there is already a proposed fix, so you can try that out if you like:
https://github.com/espressif/arduino-esp32/pull/4868

closed 2021-02-28 by per1234

xavbart, 2021-03-01:
Thanks. My search was not good prior to raising it!

damianacatanoso, 2021-03-22:
Hi xavbart, I had the same issue. I had downloaded my code from a google drive folder, compressed in .zip. After download, Mac automatically extracted my files in the "download" folder. Running the code in that folder gave me the exit status 64 error. I just copied and pasted the code outside of that folder, say Desktop, and it compiled just fine. I suggest you try relocating your code somewhere else and give it a try!

xavbart, 2021-03-23:
> Hi @xavbart, I had the same issue. 
I can compile my code just fine, not to worry, moving projects to another place, it is just that the issue prevents from using normal paths on Mac and that is something that should be sorted but it is linked to espressif contribution. Hopefully will get sorted.