arduino/Arduino #11313: Arduino Library Manager: Library Name Change

lakshanthad, 2021-03-01:
Hi, 

The name for the following library, inside library.properties have been changed:

https://github.com/Seeed-Studio/Seeed_Arduino_rpcWiFi

I have already updated the version inside library.properties and created a new release as well. 

Please help to include this update in Arduino Library Manager

Best Regards,
Lakshantha

lakshanthad, 2021-03-01:
Can I know how long it will take for this update to appear on Arduino Library Manager?

Best Regards,
Lakshantha

per1234, 2021-03-01:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it eventually. Please be patient.

lakshanthad, 2021-03-01:
I understand.

Thank you for the clarification.

cmaglie, 2021-03-03:
Done.

closed 2021-03-03 by cmaglie