arduino/Arduino #11316: [Library Manager] Please include "MIDIcontroller" in the Arduino Library Manager

joshnishikawa, 2021-03-02:
https://github.com/joshnishikawa/MIDIcontroller

A library for creating Teensy MIDI controllers.
Includes easy to follow examples for implementing velocity sensitive FSR, Piezo or Capacitive Touch inputs; momentary, latch or 'trigger' type MIDI buttons (also works with Cap Touch); stable analog to MIDI conversion for potentiometers and other sensors; stable Capacitive Touch to MIDI conversion for expression control; support for encoders; Min/Max output can be set (or inverted) for all MIDI; Min/Max input can also be set for analog input (maintaining stability in conversion)

per1234, 2021-03-02:
Hi @joshnishikawa. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

joshnishikawa, 2021-03-02:
Okay. Sorry I missed a step. There's now a release and a tag with the same version number. Thanks!

cmaglie, 2021-03-03:
Done.

closed 2021-03-03 by cmaglie