arduino/Arduino #11320: Arduino download page is displaying wrong date for Arduino IDE availability

elsatch, 2021-03-02:
Whenever you try to download software from Arduino website, you are directed to the donate page. In that donation page, it says "Since its first release in March 2015, the Arduino IDE has been downloaded..."

[image]

However, the release date is incorrect. Checking the github repository, it seems like the **Arduino IDE is available at least from 2005 instead of 2015**.

afzalkhan210, 2021-03-08:
Hi I would like to work on this Issue

per1234, 2021-03-27:
I see the text has since changed:
>Since the release 1.x release in March 2015

I see there was a 1.6.1 and a 1.6.2 release in March 2015. I guess it's referring to one of the two.

Since there is still a typo in that text (a surplus "release"), I'll leave this open.

afzalkhan210, 2021-03-28:
Hi,I'm new to open source may I try to fix this please

per1234, 2021-03-28:
@afzalkhan210 unfortunately the source code for the website is not stored in a public repository so this will be impossible.

If you want to make a contribution to the Arduino project then that is welcome, but asking about it here on the issue tracker is not the right approach. Just find an issue you know you can fix and fix it. You don't need to ask permission. If you need assistance with the work, then post on the Arduino Forum. That is the appropriate place:
https://forum.arduino.cc/

afzalkhan210, 2021-03-29:
Thank you it will be my pleasure and sorry I didn't knew about it before