arduino/Arduino #11322: Fritzing model for Nano Every has wrong footprint

RudolfAtHome, 2021-03-03:
Official Arduino Nano Page, Documentation section, Fritzing model:
The Breadboard item is a little to large. It should have a pin distance of 0,1 inch.
The other elements are OK.

wmhoneycutt, 2021-06-30:
Are we sure that the PCB footprint is OK? I noticed that the Pin spacing is fine horizontally, but vertically, the pins are spaced less than 0.1"

BastianFie, 2022-04-19:
Could it be that the issue (reported [here](https://forum.arduino.cc/t/fritzing-arduino-nano-every-wrong-model-in-the-fritzing-download-section/683725)) is fixed now? I just try to play with the Arduino Nano Every and it looks fine to me (using Fritzing 0.9.9):

[image]

I downloaded the file (https://docs.arduino.cc/static/ff3f453bf48a903e37f9f66e11861b36/fritzing.fzpz) from https://docs.arduino.cc/hardware/nano-every