arduino/Arduino #11324: [Library Manager] Add Seeed Arduino rpcBLE to Arduino Library Manager

lakshanthad, 2021-03-04:
Hi,

Please add the following library to Arduino Library Manager:

https://github.com/Seeed-Studio/Seeed_Arduino_rpcBLE

Best Regards,
Lakshantha

lakshanthad, 2021-03-16:
Hi, 
Can I know whether it will take longer for the library to get added?

Best Regards,
Lakshantha

per1234, 2021-03-16:
Hello @lakshanthad 

Your library will not be available for installation via Library Manager until it has been added to the index. A maintainer occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

lakshanthad, 2021-03-17:
okay. Thank you

cmaglie, 2021-03-21:
Done, sorry for the long delay!

closed 2021-03-21 by cmaglie

lakshanthad, 2021-03-21:
It's fine. Thank you