arduino/Arduino #11325: [Library Manager] Add esp_moga library to Arduino Library Manager

sensboston, 2021-03-04:
Could you please add my new library for MOGA bluetooth gamepads support for ESP32 (probably, it also will work with Arduino BT shields - any device working with **BluetoothSerial.h** - but I don't have one to test)

https://github.com/sensboston/esp32_moga

Thank you!

per1234, 2021-03-04:
Hi @sensboston. Thanks for your interest in contributing to the Arduino Library Manager index.

In order for your library to be added to the index, you will need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

sensboston, 2021-03-04:
Done: https://github.com/sensboston/esp32_moga/releases

per1234, 2021-03-04:
>I hope, I've done well

Looks fine to me!

>Could you specify exactly how this should be done

The truth is that it really doesn't matter what the tag name is. At the time I wrote the stock reply above I didn't understand as well how the indexer worked. However, I do think it's best practices for the tag name to reflect the version number in some manner, so I've left it like that. I apologize if it caused you some confusion. Some people like to use the "v" prefix on release tag names, and GitHub even seems to encourage this practice, some people just use the version number alone. I don't think it really matters so long as you are consistent.

closed 2021-03-21 by cmaglie