arduino/Arduino #11328: option to reload /dev/tty* entries

petaflot, 2021-03-05:
in certain conditions the available "Serial port" entries are out-of date ; it is therefore required to completely quit the IDE so that the list is up-to-date

suggestion : add a "reload available serial ports" entry to to the "Tools > Port" submenu called sth like "refresh serial ports list"

per1234, 2021-03-05:
Hi @petaflot. Thanks for your report. Have you tried Arduino IDE 2.0.0-beta.3 to see if this issue still occurs? If not, please try it out and then let us know what you found. You can download it here:
https://www.arduino.cc/en/software#experimental-software

petaflot, 2021-03-05:
nope, haven't tried it yet. have you tried making a dev-embedded/arduino-9999 ebuild (gentoo)?

matthijskooijman, 2021-03-06:
> in certain conditions the available "Serial port" entries are out-of date ; it is therefore required to completely quit and restart the IDE so that the list is up-to-date

That actually surprises me, since AFAIR the IDE 1.8.x code reloads the serial port list every time the tools (or ports, don't remember exactly) menu is reloaded.

You suggest you are using the Gentoo-build, I wonder if they changes something in this area. In addition to trying this with 2.x, you could also try downloading the 1.8.x zipfile and try that (no installation required), to check if this is maybe a problem in the Gentoo version only.

petaflot, 2021-03-06:
only a single patch is applied:

-    loadLib(new File(Base.getContentFile("lib"), System.mapLibraryName("astylej")));
+    System.loadLibrary("astylej");

in arduino-core/src/processing/app/Platform.java and app/src/cc/arduino/packages/formatter/AStyleInterface.java

comment in ebuild mentions:
# We need to load system astyle/listserialportsc instead of bundled ones.

per1234, 2021-03-27:
Closing as off topic for this repository. Arduino only supports the official builds of the Arduino IDE. If you are able to reproduce the issue with the official Arduino IDE then please comment here and I'll reopen.

closed 2021-03-27 by per1234