arduino/Arduino #11349: [library manager] Please add the following  library

PhotonBoards, 2021-03-14:
Please add my library:
https://github.com/PhotonBoards/Photon

Thank you!

cmaglie, 2021-03-21:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

per1234, 2021-04-17:
Closing due to lack of action to bring the library into compliance with the requirements for addition to the Library Manager index.

@PhotonElectronic if you ever get around to taking the required action, you can comment here and I'll be happy to re-open this issue.

closed 2021-04-17 by per1234