arduino/Arduino #11357: PCF8575 by Mischianti Renzo

gerryr1gerryr1, 2021-03-18:
First post. Hope this is the right place.

Using Arduino Mega 2560 had problems getting this Library to work.

Using the PCF8575 Module ( which has no output pins 8 or 9),the outputs were

OK on Pins 0 to 7 but the pins 10 to 17 were out by two places.

eg. sketch code p8 activated Pin 10. Sketch code p9 activated Pin 11 etc etc.

With the help of the Arduinp forum the problem appears to be in the  .h file.

Changed the #define lines in the .h file to;

#define P00   0
#define P01   1
#define P02   2
#define P03   3
#define P04   4
#define P05   5
#define P06   6
#define P07   7
#define P10   8
#define P11   9
#define P12   10
#define P13   11
#define P14   12
#define P15   13
#define P16   14
#define P17   15

#include <math.h>

Changing the code to the above fixed the problem.

Thanks to Arduino Forum's PaulRB for the fix.

gerryr1gerryr1, 2021-03-18:
Sorry Forget to mention

Library is PCF8575 Library by Renzo Mischianti Ver 1.0.1

per1234, 2021-03-18:
Hi @gerryr1gerryr1. This issue tracker is only to be used to report bugs or feature requests for official projects of the Arduino organization. This library is written and maintained by a 3rd party member of the Arduino community, not by the Arduino organization so your report is off topic here.

closed 2021-03-18 by per1234