arduino/Arduino #11359: Library release not updating: RevEng PAJ7620 - v1.5.0

acrandal, 2021-03-19:
Dear Arduino IDE/CLI library team:

A while back (around December) you awesomely got my RevEng PAJ7620 library added to the searchable libraries. The releases are available on GitHub (https://github.com/acrandal/RevEng_PAJ7620). The initial release was 1.3.0 and that worked great. When I issued the 1.4.0 release it updated on the IDE downloads great, but since then my 1.4.1 and 1.5.0 releases haven't updated in the official system.

Is there something else I should do to get the new library version available in the IDE?

per1234, 2021-03-19:
Hi @acrandal you can see from this screenshot that 1.5.0 is available for me:
[image]
Do you still not have 1.5.0 available in your Library Manager?

I also don't have access to 1.4.1 and I was not able to spot any reason why that tag should have been rejected by the indexer. Our Library Manager wizard, Cristian will surely be able to say what is happening.

acrandal, 2021-03-19:
@per1234 - I do have 1.5.0 in my IDE at this point. I have no idea what might have gone wrong there (probably me along the way). Thank you for checking on it. How often does the indexer run, so I know how long to wait before suspecting a possible issue to investigate?

Knowing why the 1.4.1 release isn't showing up would be good. If I have to issue any more bugfixes (not that I ever write code with a flaw) it would be nice to be able to use that last part of the version number for it. Hopefully Christian will have some insights.

You and your team do great work. Thank you, I truly appreciate it.

cmaglie, 2021-04-02:
For some reason, the local git clone on the indexer is not able to fetch that particular tag `1.4.1` from github:

```
github.com/acrandal/RevEng_PAJ7620$ git tag
docs-deploy-v1.4.0
docs-deploy-v1.4.1
docs-deploy-v1.5.0
v1.2.0
v1.3.0
v1.4.0
v1.5.0
github.com/acrandal/RevEng_PAJ7620$ git fetch --all --tag
Fetching origin
github.com/acrandal/RevEng_PAJ7620$ git tag
docs-deploy-v1.4.0
docs-deploy-v1.4.1
docs-deploy-v1.5.0
v1.2.0
v1.3.0
v1.4.0
v1.5.0
```

BTW if I try to clone a fresh copy the tag `v1.4.1` is there.
It's the first time that I see something like this and, honestly, I really don't know how this is possible, for some reason the local git clone on the server got messed up.

To fix this up I removed the old git clone and let the server download it again, and now it's working as expected.
I kept a backup copy of the corrupted git clone, just in case this will happen again, but I will not investigate further for now...

closed 2021-04-02 by cmaglie