arduino/Arduino #11361: NanoBLE33Sensor - arm_rms_q15  type not found by linker

as93, 2021-03-19:
### /home/andrew/Arduino/libraries/Nano33BLESensor/src/Nano33BLEMicrophoneRMS.cpp:105: undefined reference to `arm_rms_q15' ###
I checked source code in the library arm_math.h and the definition is there 
**linker output ->**
home/andrew/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-gcc -c -Os -g -nostdlib @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/cflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -DARDUINO=10813 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino/api/deprecated -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE -I/home/andrew/Arduino/libraries/ArduinoBLE/src -I/home/andrew/Arduino/libraries/Nano33BLESensor/src -I/home/andrew/Arduino/libraries/Arduino_LSM9DS1/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/Wire -I/home/andrew/Arduino/libraries/Arduino_APDS9960/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src -I/home/andrew/Arduino/libraries/Arduino_LPS22HB/src -I/home/andrew/Arduino/libraries/Arduino_HTS221/src -iprefix/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/includes.txt -o /tmp/arduino_build_398758/libraries/PDM/stm32/audio.c.o /home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src/stm32/audio.c
/home/andrew/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-gcc -c -Os -g -nostdlib @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/cflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -DARDUINO=10813 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino/api/deprecated -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE -I/home/andrew/Arduino/libraries/ArduinoBLE/src -I/home/andrew/Arduino/libraries/Nano33BLESensor/src -I/home/andrew/Arduino/libraries/Arduino_LSM9DS1/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/Wire -I/home/andrew/Arduino/libraries/Arduino_APDS9960/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src -I/home/andrew/Arduino/libraries/Arduino_LPS22HB/src -I/home/andrew/Arduino/libraries/Arduino_HTS221/src -iprefix/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/includes.txt -o /tmp/arduino_build_398758/libraries/PDM/stm32/wm8994.c.o /home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src/stm32/wm8994.c
/home/andrew/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-gcc -c -Os -g -nostdlib @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/cflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -DARDUINO=10813 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino/api/deprecated -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE -I/home/andrew/Arduino/libraries/ArduinoBLE/src -I/home/andrew/Arduino/libraries/Nano33BLESensor/src -I/home/andrew/Arduino/libraries/Arduino_LSM9DS1/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/Wire -I/home/andrew/Arduino/libraries/Arduino_APDS9960/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src -I/home/andrew/Arduino/libraries/Arduino_LPS22HB/src -I/home/andrew/Arduino/libraries/Arduino_HTS221/src -iprefix/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/includes.txt -o /tmp/arduino_build_398758/libraries/PDM/stm32/irq.c.o /home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src/stm32/irq.c
/home/andrew/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -g -Os -nostdlib @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -DARDUINO=10813 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE -I/home/andrew/Arduino/libraries/ArduinoBLE/src -I/home/andrew/Arduino/libraries/Nano33BLESensor/src -I/home/andrew/Arduino/libraries/Arduino_LSM9DS1/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/Wire -I/home/andrew/Arduino/libraries/Arduino_APDS9960/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src -I/home/andrew/Arduino/libraries/Arduino_LPS22HB/src -I/home/andrew/Arduino/libraries/Arduino_HTS221/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino/api/deprecated -iprefix/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/includes.txt /home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src/stm32/PDM.cpp -o /tmp/arduino_build_398758/libraries/PDM/stm32/PDM.cpp.o
/home/andrew/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -g -Os -nostdlib @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -DARDUINO=10813 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE -I/home/andrew/Arduino/libraries/ArduinoBLE/src -I/home/andrew/Arduino/libraries/Nano33BLESensor/src -I/home/andrew/Arduino/libraries/Arduino_LSM9DS1/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/Wire -I/home/andrew/Arduino/libraries/Arduino_APDS9960/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src -I/home/andrew/Arduino/libraries/Arduino_LPS22HB/src -I/home/andrew/Arduino/libraries/Arduino_HTS221/src -I/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino/api/deprecated -iprefix/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/cores/arduino @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/includes.txt /home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM/src/utility/PDMDoubleBuffer.cpp -o /tmp/arduino_build_398758/libraries/PDM/utility/PDMDoubleBuffer.cpp.o
Compiling library "Arduino_LPS22HB"
/home/andrew/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -g -Os -nostdlib @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -mfloat-ab [... cut ...]
Using library Wire in folder: /home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/Wire (legacy)
Using library Arduino_APDS9960 at version 1.0.3 in folder: /home/andrew/Arduino/libraries/Arduino_APDS9960 
Using library PDM at version 1.0 in folder: /home/andrew/.arduino15/packages/arduino/hardware/mbed/1.3.2/libraries/PDM 
Using library Arduino_LPS22HB at version 1.0.0 in folder: /home/andrew/Arduino/libraries/Arduino_LPS22HB 
Using library Arduino_HTS221 at version 1.0.0 in folder: /home/andrew/Arduino/libraries/Arduino_HTS221 
exit status 1
Error compiling for board Arduino Nano 33 BLE.

per1234, 2021-03-20:
Looks like some information about it here: https://github.com/DaleGia/Nano33BLESensor/issues/6

facchinm, 2021-03-22:
@as93  we are aware of the issue since also Tensorflow has been affected. In fact, `mbed` recent releases kept the CMSIS-DSP headers but removed the actual code, so `arm_*` math functions fail at link time. 
One solution is to import the needed file from https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP/Source/StatisticsFunctions and guard their compilation on the actual core version.