arduino/Arduino #11362: [Library Manager] Please add the Vertorix VT1100 Mini and VT1300 Shield Libraries

VertorixAU, 2021-03-19:
Please add the Vertorix VT1100 Mini and VT1300 Shield Libraries to the Library Manager:

[https://github.com/VertorixAU/Vertorix_VT1100_Mini](https://github.com/VertorixAU/Vertorix_VT1100_Mini)
[https://github.com/VertorixAU/Vertorix_VT1300_Shield](https://github.com/VertorixAU/Vertorix_VT1300_Shield)

Thanks
Michael

per1234, 2021-03-20:
Hi @VertorixAU. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

VertorixAU, 2021-03-20:
Hi @per1234 .  I have created a release for each repository.  Thanks for your help!

cmaglie, 2021-03-21:
Done.

closed 2021-03-21 by cmaglie

njh, 2021-05-27:
Hi @VertorixAU - I have noticed that these two repos have now been deleted.

Do you have any information about why?

VertorixAU, 2021-05-27:
HiI have made them private, sorry I work full time and was going to contact you on the weekend.I was under the impression Zigbee was a open standard when I made the libraries, and have found out it isn't so straight forward.  So I wish to make the libraries private now.  RegardsMichael

alexair777, 2022-11-14:
Hi, I am sorry to hear that VertorixAU. Is it a license issue?
Is there a possibility to get your libraries nevertheless?

Best regards
Alex