arduino/Arduino #11366: Updated ICOM 746 CAT Control library to V1.0.1

kk4das, 2021-03-21:
Minor update correcting the URL in library,properties.  New release is V 1.0.1

https://github.com/kk4das/IC746CAT

Thank you,

Dean Souleles

hsaturn, 2021-03-21:
As far as I understand, you don't have to ask. This'll be automatically done if your repo has a new release and if lint tools are ok.

cmaglie, 2021-03-21:
@kk4das you tagged 1,0.1 but you didn't update the version in library.properties

https://github.com/kk4das/IC746CAT/blob/main/library.properties#L2

to fix this:

1. update the version in library.properites to 1.0.1
2. commit & push to `main`
3. delete the tag 1.0.1 and redo it on the new `main`

after doing the above, the new release of the library will be published automatically within an hour with the URL updated.
I'm going ahead and closing this issue, if you need further assistance feel free to comment here.

closed 2021-03-21 by cmaglie

kk4das, 2021-03-21:
Done.

 

Thank you.

 

So after a new release I don’t need to open an issue then?

 

Thank you,

 

Dean

 

https://github.com/kk4das/IC746CAT/blob/main/library.properties#L2

per1234, 2021-03-21:
@kk4das correct. The indexer will automatically pick up every tag that is not compliant with the requirements listed here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

The indexer job runs periodically. I don't know exactly what the maximum delay between making a release and seeing it in Library Manager would be, but I'd say it should always happen within a day. If not, check the library to see if there is something in violation of the requirements listed at the link above.