arduino/Arduino #11369: Library seems to have dropped off the Managed Libraries list

MajicDesigns, 2021-03-22:
The MD_MIDIFile library at [](https://github.com/MajicDesigns/MD_MIDIFile) seems to have dropped off the managed libraries list and I don't understand why. Is there something that needs fixing for it to be re-included in the list?

per1234, 2021-03-23:
Hi @MajicDesigns. I did a search of the issues in this repository and didn't find any record of that library being submitted to the Library Manager index:
- https://github.com/arduino/Arduino/issues?q=is%3Aissue+MD_MIDIFile
- https://github.com/arduino/Arduino/issues?q=is%3Aissue+author%3AMajicDesigns

Is it possible that it was never in the index to begin with? If so, then the library can just be added as usual. If it was in the index and then disappeared from it, then there should be an investigation into the cause. It would be helpful if you could provide the link to the issue for the original submission of the library to the index.

MajicDesigns, 2021-03-23:
I can't find it in my list of issues either, but it would have been around 2017 when a whole lot of my libraries got added.

No problem, I will add a new request for adding this library into the Manager.

closed 2021-03-23 by MajicDesigns