arduino/Arduino #11372: [library manager] Please the following EEPROMTyped library to library manager.

tgolla, 2021-03-24:
https://github.com/tgolla/EEPROMTyped

tgolla, 2021-03-31:
Just wandering what the ETA is on getting the added.

cmaglie, 2021-04-02:
Done, sorry for the long delay!

closed 2021-04-02 by cmaglie

tgolla, 2021-04-05:
Receiving the following error when trying to install with Library Manager.  I've been able to duplicate the error on two different PCs running Windows 10.  Further research using the URL listed produces an "Error 522 Ray ID: 63aeb167d6f7e1be • 2021-04-05 00:35:12 UTC Connection timed out" indicating "The initial connection between Cloudflare's network and the origin web server timed out."

 Error downloading https://downloads.arduino.cc/libraries/github.com/tgolla/EEPROMTyped-1.0.0.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://downloads.arduino.cc/libraries/github.com/tgolla/EEPROMTyped-1.0.0.zip
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:248)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://downloads.arduino.cc/libraries/github.com/tgolla/EEPROMTyped-1.0.0.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.libraries.LibraryInstaller.performInstall(LibraryInstaller.java:155)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:125)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:117)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:239)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 522
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:239)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
	... 7 more

tgolla, 2021-04-05:
@cmaglie Please reference previous comment.

cmaglie, 2021-04-07:
I tried it now and it installed normally, it was a temporary down of the download server I guess.
May you retry now?

tgolla, 2021-04-07:
It is work now.  Thanks for looking in to this.