arduino/Arduino #11375: [library manager] Please the following GCodeParser library to library manager.

tgolla, 2021-03-25:
https://github.com/tgolla/GCodeParser

tgolla, 2021-03-31:
Just wandering what the ETA is on getting the added.

per1234, 2021-04-01:
Hi @tgolla. Your library will not be available for installation via Library Manager until it has been added to the index. The maintainer in charge of the index occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done.

So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache.

So we'll get to it soon, please be patient.

cmaglie, 2021-04-02:
Done! sorry for the long delay!

closed 2021-04-02 by cmaglie