arduino/Arduino #11378: new library

leresteux, 2021-03-26:
https://github.com/leresteux/BOF

BOF (Bling_On_ofF)

Arduino library for playing with relay, LED, and other On/OFF.
Description

BOF (Bling_On_ofF) is an Arduino library for relay, LED, and other On/OFF thing.

Void for the moment :

    on() to turn on the relay
    off() to turn off
    bling(float offduration,float onduration,float offduration)
    fade(float) for a fadein the float must be >1 ; for fadeout this float must be between 1 and 0
    isEnable()
    isDisable()
    isPause()
    BPM()
    process() must be call all time

per1234, 2021-03-26:
Hi @leresteux. Thanks for your interest in contributing to the Arduino Library Manager index.

I think you have neglected to test your library. Once installed, it completely breaks all compilations in the Arduino IDE:
```
loading libs from C:\Users\per\Documents\Arduino\libraries: loading library from C:\Users\per\Documents\Arduino\libraries\BOF-main: loading library.properties: Error reading file (C:\Users\per\Documents\Arduino\libraries\BOF-main\library.properties:6): Invalid line format, should be 'key=value'
```
It will not be allowed in to the Arduino Library Manager index in this state. Please make your library.properties file compliant and then either create a new release or else redo the existing release.

Once that's done, please comment here so we'll know we can proceed.

leresteux, 2021-03-27:
hello,

i have chance in library.properties file ...it was the problem. it s ok now
i have clean the examples file in the same time.

i m just testing the .zip from https://github.com/leresteux/BOF and :
---
Le croquis utilise 4862 octets (15%) de l'espace de stockage de programmes. Le maximum est de 32256 octets.
Les variables globales utilisent 336 octets (16%) de mémoire dynamique, ce qui laisse 1712 octets pour les variables locales. Le maximum est de 2048 octets.
---
it s ok

per1234, 2021-03-27:
@leresteux even though the problem is now fixed, the Arduino Library Manager index works from tags. At the time of the tag the library was still invalid;
https://github.com/leresteux/BOF/blob/1.0/library.properties#L7

Please either update the `version` value in your library.properties and then create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing tag.

Once that's done, please comment here so we'll know we can proceed.

leresteux, 2021-03-27:
i am very sorry, i am not confortable with github :/

i hope now all is ok 
https://github.com/leresteux/BOF/blob/v1.0/library.properties

leresteux, 2021-03-29:
https://github.com/leresteux/BOF/releases/tag/v1.0

https://github.com/leresteux/BOF

cmaglie, 2021-04-02:
Done.

closed 2021-04-02 by cmaglie