arduino/Arduino #11380: [Library Manager] Library for Engineer_Regulator

EngineeringRoom, 2021-03-27:
https://github.com/EngineeringRoom/Engineer_Regulator

EngineeringRoom, 2021-03-29:
Good day! I wanted to know how long to wait for a library to be made available through the Library Manager? Or did I do something wrong?

per1234, 2021-03-29:
Hi @EngineeringRoom.
>I wanted to know how long to wait for a library to be made available through the Library Manager?

Your library will not be available for installation via Library Manager until it has been added to the index. The maintainer in charge of Library Manager submissions occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

>Or did I do something wrong?

I gave it a look and it looks perfect to me so you should be all set for inclusion in the index on the next pass.

cmaglie, 2021-04-02:
Done.

closed 2021-04-02 by cmaglie