arduino/Arduino #11383: Serial monitor display clear option on upload no longer works

lensman11, 2021-03-28:
I have noticed that this option no longer seems to work since a recent update [my current version is 1.8.13].
Does anyone else know if this has been resolved in recent releases as the checkbox does nothing on my current version.
regards Colin.

per1234, 2021-03-28:
>the checkbox does nothing

Hi @lensman11. Which checkbox exactly are you referring to?

lensman11, 2021-03-29:
Sorry for the mis-information. There is a checkbox bottom right of serial monitor window to clear display which works ok but in previous versions of the IDE a clear function was issued when each compile happened!

It is this function which no longer appears to work.

I have noted some previous posts relating to similar issues..

Regards colin

per1234, 2021-03-30:
I guess you must be using a 3rd party version of the Arduino IDE because AFAIK the official Arduino Serial Monitor does not have a checkbox:

[image]

As for a change in the clearing behavior, this was intentional: https://github.com/arduino/Arduino/pull/9644

closed 2021-03-30 by per1234