arduino/Arduino #11388: Arduino crashes after removing boards from /hardware

mrguen, 2021-03-29:
Hello,

The Arduino IDE (1.8.13)  aborts launching after I  defined some boards variants in the /hardware folder and removed them.

Here is what I did

Added new variants in the /hardware folder
The variant was listed but an error showed at compilation "Error resolving FQBN: getting"
So I removed the files in /hardware
Then the Arduino IDE does not launch

It seems Arduino still has a reference to these boards but pointing to nothing. The debug shows

"
...
TRACE StatusLogger Using default SystemClock for timestamps.
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
TRACE StatusLogger Using DummyNanoClock for nanosecond timestamps.
DEBUG StatusLogger Reconfiguration complete for context[name=1e6f5c3] at URI jar:file:/C:/Program%20Files%20(x86)/Arduino/lib/pde.jar!/log4j2.xml (org.apache.logging.log4j.core.LoggerContext@f2bbc1) with optional ClassLoader: null
DEBUG StatusLogger Shutdown hook enabled. Registering a new one.
DEBUG StatusLogger LoggerContext[name=1e6f5c3, org.apache.logging.log4j.core.LoggerContext@f2bbc1] started OK.
Preparing boards...
java.lang.NullPointerException
        at processing.app.Base.rebuildProgrammerMenu(Base.java:1705)
        at processing.app.Base.<init>(Base.java:286)
        at processing.app.Base.main(Base.java:150)
"

How can I fix this?

per1234, 2021-03-29:
Hi @mrguen. When you say this:

>So I removed the files in /hardware

Are you referring to the `hardware` subfolder of the Arduino IDE installation folder?

matthijskooijman, 2021-03-29:
Can you try the "Hourly build" to see if the issue also exists there? I saw this same (or similar) issue and fixed it in git master with #10922, but that's not part of the 1.8.13 release yet.

mrguen, 2021-03-29:
@per1234 : I mean the personal /hardware folder, not in the installation path.

mrguen, 2021-03-29:
@matthijskooijman 

yes, it fixed the problem.

matthijskooijman, 2021-03-29:
Thanks for checking, then I'm going to close this as fixed.

closed 2021-03-29 by matthijskooijman

mrguen, 2021-03-30:
@matthijskooijman 

I have a similar issue now.

I could upload sketches to Atmega1284p board using the MighyCore, but now when trying to upload the same sketch for an Atmega644pa, I get the error 
"Error resolving FQBN: getting
Error compiling for board ATmega644."

matthijskooijman, 2021-03-30:
Hm, I'm not sure I understand. If you removed stuff from hardware and that is now no longer available, that sounds like expected b behavior. But that's probably not what you mean.

Maybe your could create a new issue and describe in a bit more detail what happens, including full (verbose) compilation output?

mrguen, 2021-03-31:
@matthijskooijman  The first issue was solved in the hourly build.

I can't give you more details on the second issue. I reinstalled the regular 1.8.13 version and did not investigate further on the first issue (that does not happen after a fresh install, i.e. with an empty personal / hardware folder) and apparently don't have problem with the second one.

cmaglie, 2021-05-15:
This looks like a duplicate of #10580 and should be fixed in the latest release 1.8.15.