arduino/Arduino #11396: [Library manager] Please add my library to the library manager Component: Board/Lib Manager

SmartElecRu, 2021-04-01:
Please add the following library (SE_EEPROM) to the Arduino library manager:

https://github.com/SmartElecRu/ArduinoIDE_SE_EEPROM

Thanks in advance!

per1234, 2021-04-02:
Hi @SmartElecRu. Thanks for your interest in contributing to the Arduino Library Manager index.

The structure of your repository is not currently compliant with the requirements for inclusion in the index:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Once you have brought the repository into compliance, you'll need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

SmartElecRu, 2021-04-02:
@per1234 We have corrected all the wishes. Please check again

per1234, 2021-04-02:
Unfortunately, you have introduced another violation of the requirements while resolving the previous ones:
https://github.com/SmartElecRu/ArduinoIDE_SE_EEPROM/commit/7222b9b7bbf82f4783bd05f33554e95cd00d9279#diff-f267f9ae2f5d5d56a1dbb305e95d40d4408d9ee7f71357f1f67fb18c29b7f082R1

Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

SmartElecRu, 2021-04-02:
Ok, I hope we changed all. Please, check again :)

cmaglie, 2021-04-02:
Done.

closed 2021-04-02 by cmaglie