arduino/Arduino #11399: [Library Manager] Please add

coderfls, 2021-04-04:
Please add https://github.com/coderfls/Arduino_MultiFunctionShield to Library Manager.
Thanks and Regards.

coderfls, 2021-04-14:
Nothing happens after 10 days?
:-(

per1234, 2021-04-14:
Hi @coderfls. Your library will not be available for installation via Library Manager until it has been added to the index. the index's maintainer occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done.

So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache.

We'll surely get to it soon, please be patient.

coderfls, 2021-04-14:
> Your library will not be available for installation via Library Manager until it has been added to the index. the index's maintainer occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done.

I know this. 
But it's a little bit frustrating to wait so long. What time it takes for review? I think you start a script for testing against the rules and that's done after some seconds (if no problem).
I think you have a lot things to do but on the other hand: Arduino need the work of the volunteers and if they are unhappy...
I wait longer ;-)

cmaglie, 2021-04-14:
Done, sorry for the long delay!

> I think you have a lot things to do but on the other hand: Arduino need the work of the volunteers and if they are unhappy...

you're right, but the good news is that we are working on a system to automate this last step in the library submission!

closed 2021-04-14 by cmaglie