arduino/Arduino #11405: [Library Manager] please add ddcvcp

tttttx2, 2021-04-05:
Please add the following new library:

https://github.com/tttttx2/ddcvcp

Thanks :)

per1234, 2021-04-05:
Hi @tttttx2. Thanks for your interest in contributing to the Arduino Library Manager index.

The library.properties `name` value of your library contains a disallowed character `/`. This will prevent its acceptance into the index. Please change the `name` value to a valid value, in compliance with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

After that, either create a new release/tag or else redo the existing tag.

Once that's done, please comment here so we'll know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

tttttx2, 2021-04-05:
Sorry for that. I recreated the tag and release, hope it's all good now

cmaglie, 2021-04-14:
Done.

closed 2021-04-14 by cmaglie