arduino/Arduino #11408: [Library Manager] latest version is not updating on arduino website

fabiuz7, 2021-04-06:
Hi,
I had discovered that [here](https://www.arduino.cc/reference/en/libraries/) there is the list of libraries available in Library Manager.

Is the list of releases updated automatically? I had seen that [this library](https://www.arduino.cc/reference/en/libraries/esp-logger/) is at v1.3.0, while on Library Manager I can download v1.4.0 ([here](https://github.com/fabiuz7/esp-logger-lib) the repo).

Is the library ill-formatted?

cmaglie, 2021-04-14:
Now the page is updated to 1.4.0, could it be that the library was just published and the webpage was still not updated (it usually takes some hour for a release to appear)?

fabiuz7, 2021-04-15:
I can confirm it! However, I published the update 17 days ago, and when I found out the problem 9 days were already passed, so either there is a problem of caching or the crawler that updates the website is very slow. 

For me is fine, if it happens again I will notify you.

closed 2021-04-15 by fabiuz7