arduino/Arduino #11416: list are longer than screen height, and impossible to scroll down

fpistm, 2021-04-09:
Hi,
I have a similar issue than #11069
Arduino IDE 1.8.13 on Windows 10. Examples menu has the correct behavior but not in Tools menu.
The new STM32duino core release will provide more boards support but now list are longer than screen height, and impossible to scroll down the list like in the example menu.
[image]
[image]

I need to find a workaround for this else release is compromised.

/cc @matthijskooijman FYI

_Originally posted by @fpistm in https://github.com/arduino/Arduino/issues/11069#issuecomment-816870815_

facchinm, 2021-04-12:
@fpistm the fix should be very similar to https://github.com/arduino/Arduino/pull/6411/files (only applied to the right menu). Is it possibile to test a fix using the actual (or git) version of STM32 core?

matthijskooijman, 2021-04-12:
@facchinm That's what I thought, but when I tried that, it makes the menus show up empty. I suspect that maybe the menu scroller does not handle hidden items (since that's I think what the board menus use?) properly, or maybe the board menu code adds/removes items from the menus at the wrong time (i.e. after the scroller modifies the menu to hide the extra items when the menu is displayed, and before the scroller restore the menu when it is closed again). I haven't dug in further than this, though.

fpistm, 2021-04-12:
Note that Arduino IDE 2 Beta4 manages properly the menu.

About this comment from @matthijskooijman 

> @fpistm, I think your issue is a different one. You're using (might say abusing :-p) board options to do the actual board selection, but nobody ever considered that option menus would ever be so big that they need to be scrolled, so no scroller is created for those menus. It's quite a trivial fix, though. If you create a separate issue for this to not pollute this issue, I'll whip up a quick PR to fix it.

I would love not using the board options. If the boards.txt syntax could allow to specify at least one submenu it could ease several things and end users would have a better menu comprehension because they often do not understand they need to first select a "category" then in the option select a board in the "Board part number":

[image]

This would also ease arduino-cli usage when list all  the boards:
Currently we have:
```
$ arduino-cli board listall
Board Name                           FQBN
3D printer boards                    STM32:stm32:3dprinter
Discovery                            STM32:stm32:Disco
Electronic speed controllers         STM32:stm32:ESC_board
Eval                                 STM32:stm32:Eval
Garatronic/McHobby                   STM32:stm32:Garatronic
Generic Flight Controllers           STM32:stm32:GenFlight
Generic STM32F0 series               STM32:stm32:GenF0
Generic STM32F1 series               STM32:stm32:GenF1
Generic STM32F2 series               STM32:stm32:GenF2
Generic STM32F3 series               STM32:stm32:GenF3
Generic STM32F4 series               STM32:stm32:GenF4
Generic STM32F7 series               STM32:stm32:GenF7
Generic STM32G0 series               STM32:stm32:GenG0
Generic STM32G4 series               STM32:stm32:GenG4
Generic STM32H7 Series               STM32:stm32:GenH7
Generic STM32L0 series               STM32:stm32:GenL0
Generic STM32L1 series               STM32:stm32:GenL1
Generic STM32L4 series               STM32:stm32:GenL4
Generic STM32L5 series               STM32:stm32:GenL5
Generic STM32WB series               STM32:stm32:GenWB
LoRa boards                          STM32:stm32:LoRa
Midatronics boards                   STM32:stm32:Midatronics
Nucleo-144                           STM32:stm32:Nucleo_144
Nucleo-32                            STM32:stm32:Nucleo_32
Nucleo-64                            STM32:stm32:Nucleo_64
STM32MP1 series coprocessor          STM32:stm32:STM32MP1
```
But it would be fine to have:
`Generic STM32L4 series               STM32:stm32:GENERIC_L452RETxP`

anyway I guess this requires to update also arduino-cli.

fpistm, 2021-04-12:
> 
> 
> @fpistm the fix should be very similar to https://github.com/arduino/Arduino/pull/6411/files (only applied to the right menu). Is it possibile to test a fix using the actual (or git) version of STM32 core?

You could use this PR:  https://github.com/stm32duino/Arduino_Core_STM32/pull/1091
At least the boards.txt with the Generic L4 entry.

Note that with this PR we will support more than 500 boards entry.

PR #11425 (2021-04-30): Make Custom menus scrollable
Fixes #11416

The patch on MenuScroller.java is needed to avoid a clash between custom menus with the same label.
This behaviour artificially increases the amount of objects that the scroller will calculate.
Eg. if two boards share the same custom menu, that menu will contain the properties for both the boards (since it's parsed twice).

### All Submissions:

* [x] Have you followed the guidelines in our Contributing document?
* [x] Have you checked to ensure there aren't other open [Pull Requests](https://github.com/arduino/Arduino/pulls?q=) for the same update/ch [... cut]
commit 6c3c1c09 (2021-04-30): Make Custom menus scrollable

Fixes #11416

The patch on MenuScroller.java is needed to avoid a clash between custom menus with the same label.
This behaviour artificially increases the amount of objects that the scroller will calculate.
Eg. if two boards share the same custom menu, that menu will contain the properties for both the boards (since it's parsed twice).
closed 2021-04-30 by cmaglie