arduino/Arduino #11419: [Library Manager] Request to add new library: Arduino library for RAK14001 RGB LED with NCP5623 IC

beegee-tokyo, 2021-04-10:
Please add this library to Library Manager:

Library: https://github.com/RAKWireless/RAK14002-CAP1293-Library
Library release: https://github.com/RAKWireless/RAK14002-CAP1293-Library/releases/tag/1.0.1

New library to use NCP5623 RGB LED controller. Adapted to be used with [RAKwireless](https://docs.rakwireless.com/Product-Categories/WisBlock/) RGB module.

per1234, 2021-04-11:
Hi @beegee-tokyo. Thanks for your interest in contributing to the Arduino Library Manager index.

Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

beegee-tokyo, 2021-04-11:
@per1234 
Changed name and created new tag & release.

Sorry for the name problem. Didn't check the rules.

cmaglie, 2021-04-14:
Done.

closed 2021-04-14 by cmaglie