arduino/Arduino #11426: Please add new AnalogWrite Library for ESP32-ESP32S2 Arduino core

Dlloydev, 2021-04-14:
Repository URL:
[https://github.com/Dlloydev/ESP32-ESP32S2-AnalogWrite](https://github.com/Dlloydev/ESP32-ESP32S2-AnalogWrite)

per1234, 2021-04-14:
Hi @Dlloydev. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Dlloydev, 2021-04-14:
Ahh,, I knew there was something missing. Thanks,   ... done

cmaglie, 2021-04-14:
Done.

closed 2021-04-14 by cmaglie