arduino/Arduino #11428: Add SparkFun Spectral Sensor AS7341X - AS7341/AS7341L - Arduino Library

PPVJM, 2021-04-14:
Hello,

Please add our AS7341X (AS7341/AS7341L) library to the library manager. Here is the link:
[https://github.com/sparkfun/SparkFun_AS7341X_Arduino_Library](https://github.com/sparkfun/SparkFun_AS7341X_Arduino_Library)

Thank you!

cmaglie, 2021-04-14:
Hi @PPVJM

I've added the library to the indexer but there is a problem in the `library.properties` https://github.com/sparkfun/SparkFun_AS7341X_Arduino_Library/blob/main/library.properties#L6-L7 the `paragraph` field must be one single line, so this two lines:

```
paragraph=This library allows the user to measure light spectrum in 10 different channels: 415nm, 445nm, 480nm, 515nm, 555nm, 590nm, 630nm, 680nm, clear and near infrared.
This sensor will measure the color components in a specific target, decomposing light into these individual components. 
```

must become:

```
paragraph=This library allows the user to measure light spectrum in 10 different channels: 415nm, 445nm, 480nm, 515nm, 555nm, 590nm, 630nm, 680nm, clear and near infrared. This sensor will measure the color components in a specific target, decomposing light into these individual components. 
```

don't worry about formatting, the library manager will display it correctly.

To resolve this:

1. edit and fix the `paragraph` field in `library.properties`
2. commit & push
3. delete the release tag `1.0.0` and re-do it on the new main

once done, the library will be automatically published within an hour. I'm going ahead and close the issue, please comment here if you need further assistance.

closed 2021-04-14 by cmaglie

PPVJM, 2021-04-17:
Hi Cristian

Thank you for your email.

I have already followed your instructions by deleting both the release and
the tags and reuploading them.

I hope I did it right.

Thanks,

Ricardo Ramos