arduino/Arduino #11439: Add library request

jordancrubin, 2021-04-19:
Please add this sprinkler system library to the Library index

https://github.com/jordancrubin/sprinkler-system

Thank you

per1234, 2021-04-19:
Hi @jordancrubin. Thanks for your interest in contributing to the Arduino Library Manager index.

Although I see that your repository is currently compliant with all [the requirements for addition to the Arduino Library Manager index](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ), the indexer works from Git tags and at the time the tag was created, the repository was not compliant:
https://github.com/jordancrubin/sprinkler-system/tree/1.0.0

You need to create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file (I see you've already incremented the `version` value to 1.0.1.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

jordancrubin, 2021-04-19:
Hello i have corrected the value in lib.properties to 1.0.0

per1234, 2021-04-19:
Hi @jordancrubin. I apologize for not being clear in my instructions. Changing the `version` value in library.properties is not what was required.

Please do this:

1. Choose the version number for your next release.
1. Change the `version` value in your library.properties to that version number: https://github.com/jordancrubin/sprinkler-system/blob/master/library.properties#L3
1. Create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) for the version number you chose.
1. Comment here to let us know that you have done so.

jordancrubin, 2021-04-23:
>    2. Change the version value in your library.properties to that version
>    3. Create a new release
>    4. Comment here to let us know that you have done so.

Hello, I have conducted these steps in what is now release 1.0.1.

thanx

Jordan

jordancrubin, 2021-04-25:
Is there something still required with regard to changes?

per1234, 2021-04-25:
@jordancrubin everything looks perfect to me now. Your library will be added the next time the maintainer in charge of the Library Manager index processes all the outstanding requests.

closed 2021-04-26 by cmaglie