arduino/Arduino #11441: [Library Manager] Problems installing a library that's >5Meg on Windows 10.

crankyoldgit, 2021-04-19:
Hi, I'm the owner/maintainer of the [IRremoteESP8266 Arduino library](https://github.com/crankyoldgit/IRremoteESP8266), & I've been trying to run down a [user reported problem](https://github.com/crankyoldgit/IRremoteESP8266/issues/1451) where our latest versions for our library are not installing correctly on the Windows 10 version of the IDE (i.e. Arduino v1.8.13, inc. nightlies, etc. v2.x has not been tried.) under the built-in Library Manager. All versions _prior_ to v2.7.16 of our library (https://github.com/crankyoldgit/IRremoteESP8266) install fine. As of v2.7.16, the `IRremoteESP8266` directory under `C:\Users\USERNAME\Documents\Arduino\libraries` disappears. It reappears if v2.7.15 or lower is installed.
The IDE installs all versions of the library as expected under the Linux (x64 1.8.13) version of the Arduino IDE.

I've gone over our library with a fine tooth comb, the Arduino Library linter passes with no errors or warnings.

The only thing I can come up with is v2.7.15 (and earlier) of our library, the Arduino Library Manager zip files have a compressed size _less than_ 5 Megabytes. As of v2.7.16+ the zip files have a compressed size _larger than_ 5 Megabytes.

You can see more detail about what processes & debugging I've gone through to try to narrow down the problem in https://github.com/crankyoldgit/IRremoteESP8266/issues/1451, with the TL;DR: Summary in https://github.com/crankyoldgit/IRremoteESP8266/issues/1451#issuecomment-822159773.

In short, our library works as expected in PlatformIO, manually downloading the zip file & unpacking etc (from Github, or via the `staging` directory of the IDE, or the corresponding URL in the `library_index.json.gz` file, **and** on the Linux version of your IDE on all versions of the library. Windows 10, not so much past v2.7.15. It's been working fine, then seemingly out of the blue ...

Can you please fix this, or tell me what I'm doing wrong that has angered the Windows 10 Arduino IDE/Library Manager gods? 

P.S. Yes, I know the library is big. I'm working to reduce it's size in the meantime.

per1234, 2021-04-19:
Hi @crankyoldgit. I think the problem is the same as was reported here:
https://github.com/arduino/Arduino/issues/11059#issuecomment-817329218

Here you have two files whose paths differ only in case:
- https://github.com/crankyoldgit/IRremoteESP8266/blob/v2.7.16/docs/doxygen/html/ir__EcoClim_8cpp.html
- https://github.com/crankyoldgit/IRremoteESP8266/blob/v2.7.16/docs/doxygen/html/ir__Ecoclim_8cpp.html

(note the "C" vs "c")
These files can not co-exist on a case insensitive file system such as Windows or macOS.

`docs/doxygen/html/ir__Ecoclim_8cpp.html` was added after the v2.7.15 release,

crankyoldgit, 2021-04-19:
> These files can not co-exist on a case insensitive file system such as Windows or macOS.

Ooo. That makes sense. I'll see if that fixes it.

it would be great if the Arduino Library Linter and/or the IDE produced a visible (in a log etc) entry when this happens or is detected.

crankyoldgit, 2021-04-20:
@per1234 Thanks. I can confirm that seems to be the underlying issue that caused the "fun & games" <tm><pat. pending> we were having. Thanks for your sage advice. It would have take me a long time to track that down.

per1234, 2021-05-12:
>it would be great if the Arduino Library Linter and/or the IDE produced a visible (in a log etc) entry when this happens or is detected.

This is an excellent suggestion! I have put it on the "to do" list. My thought is that probable the best approach is to add a check to the linter, blocking any release with this issue from being added to the Library Manager index, with a clear error message in the logs. A message from the IDE would have been useful in this case, but that's only exposing the problem once it has reached the user, who is less able to do anything about it. It's better for the issue to be caught by the library maintainer before it gets to the user.

I do notice that Arduino CLI and Arduino IDE 2.0.0-beta.6 are able to install IRremoteESP8266@2.7.16 on Windows without any problem. So it won't be so much of a problem in the future. However, they don't provide any communication about the file having been overwritten. Even though it wouldn't matter in this particular case, there might be other situations where the overwritten file was needed.

As for the issue with IRremoteESP8266 as it stands now, I see it has been resolved in the 2.7.18 release so I'll go ahead and close this. If you decide you want to remove the problematic releases from the Library Manager index, you can follow the procedure documented here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#sorry-i-did-something-wrong-how-can-i-change-or-unpublish-an-already-published-library

closed 2021-05-12 by per1234