arduino/Arduino #11449: please add DBS-Lib 0.0.2

DBSStore, 2021-04-21:
https://github.com/DBSStore/DBS_Lib/releases/tag/0.0.2
https://github.com/DBSStore/DBS_Lib

per1234, 2021-04-21:
Hi @DBSStore. The Library Manager indexer checks all the libraries on the list for new tags/releases hourly. After that, it may take some time for the updated index to propagate through the CDN cache to make your new release available everywhere, but it will get there in good time.

So you don't need to create an issue report every time you release a new version; once you push a tag/release that is compliant with all the Library Manager requirements everything happens automagically from there.

Unfortunately, I see that your 0.0.2 release is not compliant. The problem is that the `version` field value in library.properties was not updated before making the 0.0.2 release:
https://github.com/DBSStore/DBS_Lib/blob/0.0.2/library.properties#L2
Although the unit of the Library Manager index is the Git tag, the tag name itself is irrelevant to the indexer. The sole source of versioning information is the library.properties `version` field. So any tag that has the same `version` value as a tag that is already in the index will not be added to the index by the indexer.

So you'll need to update the `version` value and then either create a new tag or else recreate the existing tag. After that, the release will be picked up automatically.

Reference:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

closed 2021-04-21 by per1234

DBSStore, 2021-04-22:
https://github.com/DBSStore/DBS_Lib/releases/tag/0.0.2 

I still can't edit and update.

per1234, 2021-04-22:
You still haven't updated the `version` value in library.properties, so it's no surprise that you're getting the same result:
https://github.com/DBSStore/DBS_Lib/blob/0.0.2/library.properties#L2

DBSStore, 2021-04-22:
Please check it, I fixed it.