arduino/Arduino #11466: Please add my library SIKTEC-SdExplore to the library manager

siktec-lab, 2021-04-30:
Please add my library [SIKTEC-SdExplore](https://github.com/siktec-lab/SIKTEC-SdExplore)
A companion lib to SdFat which handles scanning and searching files on an SD Card.

per1234, 2021-04-30:
Hi @siktec-lab. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

siktec-lab, 2021-04-30:
Sorry for that -> Release pushed 1.0.2

cmaglie, 2021-05-05:
Done.

closed 2021-05-05 by cmaglie