arduino/Arduino #11469: Please add my library IoT_Modules-Buttons to the library manager.

gq97a6, 2021-04-30:
Please add my library to library manager.

https://github.com/MartinXBcn/ESP32Logger

Best regards, gq97a6.

cmaglie, 2021-05-05:
@gq97a6 are you sure the link is correct?

gq97a6, 2021-05-06:
@cmaglie Whoopsie. Nope, its not. Correct link: https://github.com/gq97a6/IoT_Modules-Buttons

per1234, 2021-05-06:
Hi @gq97a6. I took a look at the https://github.com/gq97a6/IoT_Modules-Buttons library and spotted a problem:

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

gq97a6, 2021-05-06:
@per1234 Done.

cmaglie, 2021-05-11:
Done.

closed 2021-05-11 by cmaglie