arduino/Arduino #11470: Incorrect SPI pins in Leonardo pinout.

oqibidipo, 2021-05-01:
[https://content.arduino.cc/assets/Pinout-Leonardo_latest.pdf](url) 
pages 1 and 2

Pins 10 to 13 are marked as SPI pins when in reality SPI is only on the ISP header.

FunzioneFabio, 2021-07-01:
Thank you @oqibidipo for poiting out this error, the pinout is now corrected and online!
https://content.arduino.cc/assets/Pinout-Leonardo_latest.pdf
https://store.arduino.cc/arduino-leonardo-with-headers

closed 2021-07-01 by kengdahl

gentilkiwi, 2024-08-01:
> Thank you @oqibidipo for poiting out this error, the pinout is now corrected and online! https://content.arduino.cc/assets/Pinout-Leonardo_latest.pdf https://store.arduino.cc/arduino-leonardo-with-headers

Can you also please the documentation, on https://docs.arduino.cc/language-reference/en/functions/communication/SPI/#description by example ?