arduino/Arduino #11474: [Library Manager] Request to add to Library Manager: RTT-Ethernet

onelife, 2021-05-02:
Please help to add my RTT-Ethernet library to the library manager:
[RTT-Ethernet](https://github.com/onelife/RTT-Ethernet)

Thank you!

per1234, 2021-05-02:
Hi @onelife. Thanks for your interest in contributing to the Arduino Library Manager index.

Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). I see that you have set the `name` to a unique value:
https://github.com/onelife/RTT-Ethernet/blob/master/library.properties#L1
However, the Arduino Library Manager indexer works from tags, and at the time of the latest tag, the name in use was already taken:
https://github.com/onelife/RTT-Ethernet/blob/1.2.0/library.properties#L1

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

onelife, 2021-05-03:
Hi @per1234. Thanks for the help! I have done the release.

cmaglie, 2021-05-05:
Done.

closed 2021-05-05 by cmaglie