arduino/Arduino #11475: Nano's pins A6 and A7 marked as "Digital Pins" on pinout diagram

per1234, 2021-05-02:
https://store.arduino.cc/arduino-nano

On the "Pinout Diagram", both the summary as well as the full version, pins A6 and A7 are incorrectly marked as having "Digital Pin" capabilities.

On the classic Nano, these pins only have ADC capabilities, and can not be used as digital input or outputs.

oqibidipo, 2021-05-02:
Also, under Tech Specs it says
> Digital I/O Pins    22 (6 of which are PWM)

On the other hand, for Uno it says
> Digital I/O Pins   14 (of which 6 provide PWM output)

per1234, 2021-05-02:
>Also, under Tech Specs it says

That documentation error is being tracked here: https://github.com/arduino/Arduino/issues/6313

FunzioneFabio, 2021-07-01:
Thank you Per for pointing out this, the pinout is now corrected and online!
https://content.arduino.cc/assets/Pinout-NANO_latest.pdf
https://store.arduino.cc/arduino-nano

SimonJohnCastle, 2022-02-02:
This item is not completly solved.
The wrong diagram with D20 and D21 is still showing if you go to [https://docs.arduino.cc/hardware/nano](https://docs.arduino.cc/hardware/nano) and press the PINOUT button

FunzioneFabio, 2022-02-16:
@SimonJohnCastle Thank you for pointing that out, they are now updated!
@kengdahl you can close the issue.

closed 2022-04-21 by kengdahl