arduino/Arduino #11486: Add Tuya wifi mcu-sdk library

flyingcys, 2021-05-06:
Hi :)
Could you please add this library to the library manager?
https://github.com/tuya/tuya-wifi-mcu-sdk-arduino-library

Thanks!

per1234, 2021-05-06:
Hi @flyingcys. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

flyingcys, 2021-05-06:
I have added the release version 0.0.1

Thanks!

cmaglie, 2021-05-11:
Done.

closed 2021-05-11 by cmaglie