arduino/Arduino #11488: explicit inclusion of <Arduino.h> triggers compilation error

AndrewBCN, 2021-05-06:
Hello,
I did check the open and closed issues and the closest I have found is this: https://github.com/arduino/Arduino/issues/4352
but it's not the same.

I am using Ubuntu 20.10, Arduino 1.8.13 and STM32 core 2.0.0.

In some of my sketches I test for the STM32 core version with this code
`#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  < 0x02000000)
#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"
#endif
`
This compiles totally fine.

However, if in my sketch I explicitly add:
`#include <Arduino.h>`

this triggers the compilation error in the test.

It was my understanding that the Arduino build system automatically adds the `#include <Arduino.h>` to the top of the generated cpp code, so that would be a double inclusion and shouldn't change anything at all, but it seems it does.

Some help to solve this conundrum would be very much appreciated.

Note that the example sketches in the U8g2/u8x8 library all explicitly add `#include <Arduino.h>`, that's how I stumbled upon this problem.

matthijskooijman, 2021-05-06:
> It was my understanding that the Arduino build system automatically adds the #include <Arduino.h> to the top of the generated cpp code, so that would be a double inclusion and shouldn't change anything at all, but it seems it does.

This is correct, though I think it omits this if it is already found in the sketch.

Can you provide a minimal example that triggers the problem, and provide full verbose compilation output of that?

AndrewBCN, 2021-05-06:
Yes, here is a minimal example of the problem, and thank you very much for taking a look at this puzzling problem.
1) This is an example sketch included with the U8g2 library, u8x8/HelloWorld. Note that I have commented out the #include <Arduino.h>, so this code compiles fine without triggering the error in the STM32 core version test:

#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  < 0x02000000)
#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"
#endif

// #include <Arduino.h>
#include <U8x8lib.h>

#include <Wire.h>

U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE); 	      

void setup(void)
{
  u8x8.begin();
  u8x8.setPowerSave(0);
}

void loop(void)
{
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.drawString(0,1,"Hello World!");
  delay(2000);
}

2. Now if I uncomment the #include <Arduino.h>, this triggers the error during compilation:

#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  < 0x02000000)
#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"
#endif

#include <Arduino.h>
#include <U8x8lib.h>

#include <Wire.h>
...

Here is the message I get:
HelloWorld:41:2: error: #error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"
   41 | #error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"
      |  ^~~~~
exit status 1
#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"

And the full verbose compilation output is as follows:

/opt/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.13/hardware -hardware /home/andrew/.arduino15/packages -tools /opt/arduino-1.8.13/tools-builder -tools /opt/arduino-1.8.13/hardware/tools/avr -tools /home/andrew/.arduino15/packages -built-in-libraries /opt/arduino-1.8.13/libraries -libraries /home/andrew/Arduino/libraries -fqbn=STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE,upload_method=dfuMethod,xserial=generic,usb=CDCgen,xusb=FS,opt=osstd,rtlib=nano -vid-pid=0483_5740 -ide-version=10813 -build-path /tmp/arduino_build_942542 -warnings=none -build-cache /tmp/arduino_cache_607038 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0 -prefs=runtime.tools.STM32Tools.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/STM32Tools/2.0.0 -prefs=runtime.tools.STM32Tools-2.0.0.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/STM32Tools/2.0.0 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.3.1-1.3.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3 -verbose /tmp/arduino_modified_sketch_812762/HelloWorld.ino
/opt/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.13/hardware -hardware /home/andrew/.arduino15/packages -tools /opt/arduino-1.8.13/tools-builder -tools /opt/arduino-1.8.13/hardware/tools/avr -tools /home/andrew/.arduino15/packages -built-in-libraries /opt/arduino-1.8.13/libraries -libraries /home/andrew/Arduino/libraries -fqbn=STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE,upload_method=dfuMethod,xserial=generic,usb=CDCgen,xusb=FS,opt=osstd,rtlib=nano -vid-pid=0483_5740 -ide-version=10813 -build-path /tmp/arduino_build_942542 -warnings=none -build-cache /tmp/arduino_cache_607038 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0 -prefs=runtime.tools.STM32Tools.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/STM32Tools/2.0.0 -prefs=runtime.tools.STM32Tools-2.0.0.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/STM32Tools/2.0.0 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.3.1-1.3.path=/home/andrew/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3 -verbose /tmp/arduino_modified_sketch_812762/HelloWorld.ino
Using board 'GenF4' from platform in folder: /home/andrew/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0
Using core 'arduino' from platform in folder: /home/andrew/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0
/home/andrew/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/system/extras/prebuild.sh /tmp/arduino_build_942542 /tmp/arduino_modified_sketch_812762
Detecting libraries used...
/home/andrew/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb @/tmp/arduino_build_942542/sketch/build_opt.h -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -I/tmp/arduino_modified_sketch_812762 -I/home/andrew/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/avr -I/home/andr [... cut ...]
Using library Wire at version 1.0 in folder: /home/andrew/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/Wire 
Using library SrcWrapper at version 1.0.1 in folder: /home/andrew/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/SrcWrapper 
Using library SPI at version 1.0 in folder: /home/andrew/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/libraries/SPI 
exit status 1
#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"

matthijskooijman, 2021-05-07:
It looks like you put your test *before* the `#include <Arduino.h>`, so any defines that are done inside (files included by) Arduino.h are not available before that line. Seems like it works exactly like expected: If you do *not* have a `#include <Arduino.h>`, then the IDE adds it at the top (so before your check), but if you *do* have the include, then the IDE does not add it, so your check is too early.

I think this is working as expected, so I'm closing this issue. If you feel otherwise, feel free to leave additional comments.

closed 2021-05-07 by matthijskooijman

AndrewBCN, 2021-05-07:
Thank you very much for solving this puzzle, and you are absolutely right: if I put the core version test _after_ the `#include <Arduino.h>`, the code compiles fine and does not trigger the error.