arduino/Arduino #11490: Add SparkFun KX13X Arduino LIbrary to Library Manager

edspark, 2021-05-06:
Good afternoon,

Could you please add the [SparkFun KX13X Arduino Library](https://github.com/sparkfun/SparkFun_KX13X_Arduino_Library) to the library manager at your earliest convenience?

Thank you!

per1234, 2021-05-09:
Hi @edspark. I took a look at the library and spotted a couple of problems:

---
The `version` value in library.properties is not in a semver compliant format (`v` prefix):
https://github.com/sparkfun/SparkFun_KX13X_Arduino_Library/blob/v1.0.0/library.properties#L2
I can see that this has already been fixed, but the Library Manager indexer works from tags, so the library must be compliant at the time of a tag in order to be accepted into the index.

---
Empty `url` field:
https://github.com/sparkfun/SparkFun_KX13X_Arduino_Library/blob/v1.0.0/library.properties#L8

---
Once those issues are resolved, please comment here so we'll know we can proceed.

edspark, 2021-05-10:
My apologies @per1234, both issues have (hopefully) been resolved with a new release and version number. Thank you!

cmaglie, 2021-05-11:
Done.

closed 2021-05-11 by cmaglie