arduino/Arduino #11493: Problem with library manager update: Repo name mismatch?

nseidle, 2021-05-07:
Our latest library version is not appearing in library manager. Here is the [original inclusion request](https://github.com/arduino/Arduino/issues/8799). Here is the lib [properties file](https://github.com/sparkfun/Qwiic_Capacitive_Touch_Slider_Arduino_Library/blob/master/library.properties). I can't figure out what's wrong with our library properties file and/or repo name.

@per1234 Perhaps your eagle eyes can spot what's wrong?

* I've tried renaming the name field to match the repo name
* I've done a couple releases to make sure the release matches the version inside lib.props file

per1234, 2021-05-09:
Hi @nseidle. The problem is that the library.properties `name` value at the time of the original submission to Library Manager was "SparkFun Capacitive Touch Slider - CAP1203". Since that time, it was changed to "SparkFun Qwiic Capacitive Touch Slider - CAP1203":
https://github.com/sparkfun/Qwiic_Capacitive_Touch_Slider_Arduino_Library/commit/60d0f50c75aaceedeb405dc0ab9d07b8eb54687b
And then changed again to "Qwiic Capacitive Touch Slider Arduino Library"
https://github.com/sparkfun/Qwiic_Capacitive_Touch_Slider_Arduino_Library/commit/df0e5abbfcf9406910ffaa0edba44a93200a2baa

Libraries are locked to the `name` value they used at the time of the submission. Any tag that uses a different `name` value is rejected by the indexer. The reason is that the `name` is the unique identifier for the library.

The library name can be changed on request. If you would like that to be done, just let us know. Otherwise, you can either make a new tag with the original "SparkFun Qwiic Capacitive Touch Slider - CAP1203" name value, or else redo the existing one if you prefer.

nseidle, 2021-05-09:
Thanks per1234. Please let us change it to:

name="SparkFun CAP1203 Arduino Library"

Repo has been renamed to: https://github.com/sparkfun/SparkFun_CAP1203_Arduino_Library

Library properties has been updated with a release.

cmaglie, 2021-05-11:
Done.

closed 2021-05-11 by cmaglie

nseidle, 2021-05-12:
That fixed it. Many thanks!