arduino/Arduino #11498: Documentation for stream.find() and derivatives is too sparse

pogo555, 2021-05-08:
The documentation for these functions is almost obscure. It mentions that a target string is searched for, but does not say that this is not a String object or null terminated array of char.

The documentation for Serial.find() says that characters are read from the buffer, but does not say that they are discarded.

The documentation for file.find() is not included in the documentation for the SD library. This function alters the file position, but this is not mentioned anywhere.

Every instance could be improved with a simple best use example.

alranel, 2021-05-12:
This issue should be moved to https://github.com/arduino/reference-en

per1234, 2021-05-12:
@alranel it's tricky because the report covers two different areas of the documentation.

This part is relevant to the Language Reference:
>The documentation for Serial.find() says that characters are read from the buffer, but does not say that they are discarded.

but this part is not relevant to the Language Reference:
>The documentation for file.find() is not included in the documentation for the SD library. This function alters the file position, but this is not mentioned anywhere.

kengdahl, 2021-12-21:
So what can we do about this Issue? @alranel @per1234 

I can assign the documentation to someone in Content if that would help.

per1234, 2021-12-21:
>I can assign the documentation to someone in Content if that would help.

That sounds good to me.