arduino/Arduino #11499: [Add Library to Library Manager]  for TridentTD_Linenotify V3.0.2

TridentTD, 2021-05-08:
Hi, 

Could you please update TridentTD_Linenotify V3.0.1  to  the latest V3.0.2 version in library manager?

https://github.com/TridentTD/TridentTD_LineNotify/releases/tag/3.0.2

Thank you very much.

per1234, 2021-05-08:
Hi @TridentTD. The reason your 3.0.2 release was not picked up by the Library Manager indexer is because you forgot to update the `version` value in library.properties before making the release:
https://github.com/TridentTD/TridentTD_LineNotify/blob/3.0.2/library.properties#L2
The `version` value must be unique in each tag in order to be added to the index.

closed 2021-05-08 by per1234

TridentTD, 2021-05-10:
> Hi @TridentTD. The reason your 3.0.2 release was not picked up by the Library Manager indexer is because you forgot to update the `version` value in library.properties before making the release: [...]

Thank you for checking, now fix to 3.0.2

https://github.com/TridentTD/TridentTD_LineNotify/blob/3.0.2/library.properties#L2