arduino/Arduino #11518: [Library Manager] Please add TLC5916 Lite Library

dpnebert, 2021-05-11:
Greetings,

Can you please add the following library to the registry:

https://github.com/dpnebert/TLC5916-Lite

per1234, 2021-05-11:
Hi @dpnebert. Thanks for your interest in contributing to the Arduino Library Manager index.

I took a look at your library and found that it does not currently meet the requirements for addition to the Library Manager index.

You can find the information on what is needed here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Once you have brought your library into full compliance, please comment here so we'll know we can proceed.

dpnebert, 2021-05-12:
Hello, this will be my first library submitted.  I believed I went through the checklist.  Is there an instruction you could point me towards so I can double check everything for that requirement?

per1234, 2021-05-12:
>I believed I went through the checklist.

You missed multiple things:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#is-my-git-repo-ok

>Is there an instruction you could point me towards so I can double check everything for that requirement?

The very best and easiest way to check compliance is the Arduino Lint tool:
https://arduino.github.io/arduino-lint/latest/
This is what I run on every submission request.

Arduino Lint is also valuable for ensuring continued compliance of your library so that future releases are also accepted by the automated system. For this reason, you might consider adding a GitHub Actions workflow to run Arduino Lint on your library automatically, as Arduino does for our own libraries. There is a tool that makes this very easy:
https://github.com/arduino/arduino-lint-action

dpnebert, 2021-05-12:
That is great news!  I'll work to get those things fixed and get back to
you!  Thank you!

dpnebert, 2021-05-12:
I am new at this, so I apologize in advance for being a bit difficult, and thank you for the help.  I was able to run Arduino Lint and fix the errors and warnings.

I couldn't figure out how to add the license to this repository, so I am deleting it and re-creating it.  I only found the open to add a license while creating a repository.

The URL below is different than the first URL in the initial comment.

Could you please add the following library to the registry:
https://github.com/dpnebert/TLC5916_Lite

per1234, 2021-05-12:
> I was able to run Arduino Lint and fix the errors and warnings.

I'm really happy to hear that!

>Could you please add the following library to the registry:
https://github.com/dpnebert/TLC5916_Lite

It looks very good. There is one last thing you need to do, which is to make a release. Arduino Lint doesn't check for a release because it's meant to be run before you make the release to ensure everything is good first.

You can learn how to do that here:
https://help.github.com/articles/creating-releases/

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

dpnebert, 2021-05-12:
I feel so silly, I had done that to the previous repository.  I created a release.

dpnebert, 2021-05-14:
Hello!  Sorry to bother.  I just wanted to make sure we weren't waiting on me.  I assume it takes time to get added, but I wanted to be sure I completed everything I can do in the mean time.

per1234, 2021-05-14:
Hi @dpnebert. Everything looks perfect now to me, so your work is done. The initial addition of the library to the Library Manager index is a manual process which is done in batches. The maintainer responsible for this periodically comes through and clears all the outstanding requests in a single batch. They will close this issue when that is done and GitHub will send you a notification of the issue closure. So it's only a matter of time.

dpnebert, 2021-05-14:
I had a feeling it was something like that.   Thank you for the update and have a good weekend!