arduino/Arduino #11522: Arduino 1.8.14 building issue

NicoHood, 2021-05-12:
Where can I find the download of `arduino-examples-1.9.1.zip` and **where can I find the corresponding gpg signature?**

Speaking of gpg signatures I am still searching for the arduino-cli signatures as well https://github.com/arduino/arduino-cli/issues/1265

per1234, 2021-05-12:
> Where can I find the download of arduino-examples-1.9.1.zip

Here it is:
https://github.com/arduino/arduino-examples/archive/refs/tags/1.9.1.zip

> where can I find the corresponding gpg signature?

I don't know whether this answers your question, but it is the generated asset of a "Verified" GitHub release, which uses GitHub's signature:
https://github.com/arduino/arduino-examples/releases/tag/1.9.1

cmaglie, 2021-05-12:
@NicoHood if you build with `ant` using the provided `build.xml` the integrity of the `arduino-examples-1.9.1.zip` is already verified by comparing the checksum in the file `build/arduino-examples-1.9.1.zip.sha`

```
~/Code/arduino$ cat build/arduino-examples-1.9.1.zip.sha 
edf504dcca605d34818214747c9197cb376c834d
~/Code/arduino$ curl -L https://github.com/arduino/arduino-examples/archive/1.9.1.zip -s | shasum
edf504dcca605d34818214747c9197cb376c834d  -
```

this is the same as for all the other files downloaded by the ant script `build.xml`.
I already provided the GPG signature for the source code of this repo so the files `build.xml` and `arduino-examples-1.9.1.zip.sha` should be already trusted.

cmaglie, 2021-05-12:
On a side note: providing these GPG signatures is tedious and error-prone, it's another manual step that layers on top of an already very long procedure to release the IDE. I'd like to explore an alternative using gpg-signed git tags, that are already well supported in github.

https://docs.github.com/en/github/authenticating-to-github/about-commit-signature-verification

NicoHood, 2021-05-13:
Thanks, that helped me a lot!

Signing git tags is also a very good option, on ArchLinux we also support gpg signature checks for git sources. Just make sure that the latest commit/tag is signed by the trusted key (not just the github key).

closed 2021-05-13 by NicoHood

NicoHood, 2021-05-13:
Do you know where I can download the avr libraries that are bundles by default? Someone suggested that here:
https://bugs.archlinux.org/task/67453

I'd like to package them with the avr core, that makes most sense to me I guess.

per1234, 2021-05-13:
>Do you know where I can download

Hi @NicoHood. It's all defined here:
https://github.com/arduino/Arduino/blob/1.8.14/build/build.xml#L267-L305
checksums here:
https://github.com/arduino/Arduino/tree/master/build

>I'd like to package them with the avr core

These are not AVR-specific libraries. They are compatible with multiple architectures. The AVR-specific ones are bundled with the platform already:
https://github.com/arduino/ArduinoCore-avr/tree/master/libraries

[Platform-bundled libraries](https://arduino.github.io/arduino-cli/latest/platform-specification/#platform-bundled-libraries are only accessible when compiling for a board of that platform, so bundling these libraries with Arduino AVR Boards will prevent their use for boards of other platforms. Of course, they are all in Library Manager, so the user does still have the option of installing them. However, a lot of tutorials and documentation assume that the libraries will always be present, and so omit any installation instructions.

NicoHood, 2021-05-13:
Is there a way to include the libraries with a `light_bundle`? I do not want to add the avr core, but the libraries. Could we add another build target without making it over complex?