arduino/Arduino #11523: SerialMenuCmd, a new library... perhaps !!

ARTS37, 2021-05-13:
Can you consult the following library that I developed so that if it seems interesting to you, you can add it to the registry  :

https://github.com/ARTS37/SerialMenuCmd

Best regards

Armand Rolland
armand.rolland71@gmail.com

per1234, 2021-05-13:
Hi @ARTS37. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

ARTS37, 2021-05-13:
Hi per1234,
I think I made the requested changes. I ask you to be indulgent because this is my first time with github. 

https://github.com/ARTS37/SerialMenuCmd

I remain at your disposal for any further information or new modification.

Armand

ARTS37, 2021-05-18:
hi per1234,
have you done any other remarks on library.
do you think I can submit this library on  "Project Hub, Share your work with the community"

Thank you for your assistance regarding this matter ,

Armand Rolland
armand.rolland71@gmail.com

per1234, 2021-05-19:
>have you done any other remarks on library.

Everything looks fine to me now.

>do you think I can submit this library on "Project Hub, Share your work with the community"

You're welcome to do so. I think the readers will appreciate the convenience of being able to install and update the library via Library Manager.

ARTS37, 2021-05-25:
Hi per1234,
I'm sorry to bother you again but I admit that I can't share this library on www.arduinolibraries.info or on Arduino / community / project hub. can you help me ?

Armand Rolland
armand.rolland71@gmail.com

per1234, 2021-05-25:
>I admit that I can't share this library on www.arduinolibraries.info

Since www.arduinolibraries.info is generated from the Arduino Library Manager index, your library will not appear in it until it has been added to the index.

The index's maintainer occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done.

You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. Then perhaps a little more time for www.arduinolibraries.info to update.

We'll surely get to it soon, please be patient.

As for the problem with the Arduino Project Hub, that's not clear to me. I recommend you to post on the Project Hub's dedicated category in the Arduino Forum:
https://forum.arduino.cc/c/community/project-hub/138
I monitor that category and so will be sure to see your post. I'll be happy to help you out over there.

per1234, 2021-06-07:
Your submission has now been accepted! Thanks for your contribution to the Arduino Library Manager index.

The library(s) will be available for installation via Library Manager within a day's time.

You can check the logs from the Library Manager indexer for your library(s) here:
http://downloads.arduino.cc/libraries/logs/github.com/ARTS37/SerialMenuCmd/

closed 2021-06-07 by per1234

ARTS37, 2021-06-07:
Thanks a lot for your help.
I am very honored to participate in the development of arduino.
See you soon...

Armand

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>