arduino/Arduino #11524: How can I add unofficial list of 3rd party boards support urls?

Heltec-Aaron-Lee, 2021-05-13:
I can't download 3rd party board json file anymore.
[image]

How can I add 3rd party boards json files to this list?
https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

per1234, 2021-05-13:
Thanks for your report @Heltec-Aaron-Lee 

Here's the URL:
https://resource.heltec.cn/download/package_CubeCell_index.json

I am able to reproduce the issue with Arduino IDE 1.8.14
I am not able to reproduce the issue with Arduino CLI 0.18.2:
```
arduino-cli core update-index --additional-urls https://resource.heltec.cn/download/package_CubeCell_index.json
```

I have not been able to identify what might be the cause of the problem. If I download the file and then host it on GitHub then it works fine for me.

Heltec-Aaron-Lee, 2021-05-13:
Yes, the json file in GitHub also fine for us, but the json from other servers will cause an error.

per1234, 2021-05-13:
When you say "other servers", do you have other examples of the same problem happening with different URLs, or is it limited only to https://resource.heltec.cn/download/package_CubeCell_index.json ?

Heltec-Aaron-Lee, 2021-05-14:
such as these:

https://resource.heltec.cn/download/package_CubeCell_index.json
https://resource.heltec.cn/download/package_heltec_esp32_index.json

cers000, 2025-02-17:
Now，2025-2-17 22:32:15，the 3rd party boards support urls is delete, wiki page display create new page