arduino/Arduino #11525: Updated library not available in the Library Manager

manuaatitya, 2021-05-13:
Hi, <br>
     I'm trying to use the latest version of "EASE_ArduinoCode" [(bitbucket link)](https://bitbucket.org/harmonicbionics/ease_arduinocode/src/master/) Library available from the Library Manager.  <br>
A newer version (Version 0.3.0) of the library was released 2 weeks ago and the repo's library properties were also modified to the newer version number from the older version (0.2.0). However, there's no dropdown menu available to select the correct version number of the library or to update the library and it's still in the older version as seen in the image attached below.
[image]

Kindly look into the issue as to why this is happening.

Thanks in advance.

manuaatitya, 2021-05-15:
Hi @per1234 , 
     The issue still exists and the option to update the library/choose the correct version is still not available.

per1234, 2021-05-15:
Hi @manuaatitya. Yes, I can also reproduce the issue you reported. There is some problem with the Library Manager indexer system at the moment. I apologize for any inconvenience.

The Library Manager maintainers will investigate this issue and close it once the problem has been resolved. Once that is done, your library release will automatically become available for installation via Library Manager and the problem will not reoccur on future releases.

cmaglie, 2021-05-28:
The version `0.3.0` is now published.

closed 2021-05-28 by cmaglie