arduino/Arduino #11551: Compilation fails when case of .ino file does not match case of folder it is located in

positron96, 2021-05-22:
I have a small program with 1 `.ino` file and 1 `.h` file (here it is: https://github.com/vhar/withrottle). It's for esp8266 core, but it probably doesn't matter.

The .ino file is named `WiThrottle.ino`, the header is `Config.h`.

If the project folder is named `withrottle` (lowercase), then when I click `Verify/Compile` in Arduino IDE, the IDE converts Config.h into Config.h.cpp, adds stuff from .ino file into it, which makes the order of includes wrong. `Config.h` uses types that are included in .ino file, but in resulting .cpp the contents of Config.h goes before the contents of .ino file, so compilation fails.

If the project folder is named `WiThrottle`, than compilation goes fine (the cpp file is made from .ino file, not from .h file).

Since this is happening on Windows (and Windows filesystem is case-insensitive), this looks like a hard to find and hard to fix bug.

Tested on IDE versions 1.8.13, 1.8.15, Windows x64

per1234, 2021-05-22:
Thanks for taking the time to submit an issue @positron96 .

I see we already have a prior issue report about this at https://github.com/arduino/Arduino/issues/8030.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-05-22 by per1234

positron96, 2021-05-22:
Thank you, @per1234. Somewhy that issue did not show up in search.